areaDetector/ADEiger #5: SensorThickness PV precision is too low

GDYendell, 2017-07-11:
The PREC for SensorThickness_RBV is currently 3, which means it displays 0.000 (for 0.00045m). Possibly it would be even more useful to change the unit to um.

brunoseivam, 2017-07-11:
What worries me about changing it to um is that you can potentially use the PV value as an attribute, and then this might end up in an HDF5 file downstream the plugin chain. Does NeXus require certain units or can they be specified?

Regarding the precision, so far on the CSS screen I have been displaying SensorThickness and {X,Y}PixelSize in scientific notation, but I agree with the increase to 6.

GDYendell, 2017-07-12:
It's possible something will depend on it being certain unit. It doesn't seem you can specify a unit with the attribute.

It is probably not worth any confusion it may cause in changing it. The pixel size is in m, so it may be best to have the same units for consistency.

brunoseivam, 2017-07-12:
Fixed via e44f7619c

closed 2017-07-12 by brunoseivam