areaDetector/ADEiger #6: TriggerMode selection is incorrectly mapped to reset parameter

GDYendell, 2017-08-09:
Currently the allowed_values of trigger_mode are read in and indexed by the TriggerMode PV. However, the enums are defined in a different order, so there needs to be some logic using an explicit mapping between the two.

EPICS Record:   [ 0] Internal Series [ 1] Internal Enable [ 2] External Series [ 3] External Enable
Eiger trigger_mode: ["ints",                     "exts",                     "inte",                      "exte"]

This is assuming that neither Dectris nor AreaDetector want to re-order their values, which I expect is the case.

brunoseivam, 2017-08-09:
Good catch! I guess I never triggered this bug because we only use the first and last modes here...

closed 2017-08-09 by brunoseivam

GDYendell, 2017-08-09:
Thanks, Bruno!

commit b3ec2b0f (2017-08-09): Fix Trigger Mode ordering. Fixes #6