areaDetector/ADEiger #7: Driver fails to download large data files

GDYendell, 2017-08-23:
Hi Bruno,

We are testing the frame drop rate of the internal file writer to compare to the performance issues we are seeing on the Stream interface. We are finding that if we do an acquisition of 500+ frames into a single file (NumImages and FWNImagesPerFile to 500) that the only the master file is downloaded and not the data file. The data file remains on the detector storage and the master is deleted (FWAutoRemove = Yes). We can then curl to download the data file without issue. We also found with an acquisition of 10,000 frames with 1,000 frames per file, the first 8 were downloaded OK, but the final 2 were not.

Have you seen this problem?

brunoseivam, 2017-08-23:
Hi Gary,

Yes, I saw that before, the driver was failing to download the last few files. This commit (2e17935) was supposed to solve that. Could you try enabling `ASYN_TRACE_FLOW` mask to see the sequence of events?

Could you also try to play with `MAX_RETRIES` in that bit of code to see if it helps?

My guess is that the problem stems from the FileWriter reporting it is done while still writing files.

GDYendell, 2017-08-31:
Ah, yes I remember seeing that. I didn't realise it was only on dev. We are on master (+ b3ec2b0), so it's quite likely that is the problem.

I think we have done enough testing to convince Dectris of the problem, so this won't be much of an issue for now. I will update to dev if we do need the fix.

Cheers!

closed 2017-08-31 by GDYendell