areaDetector/ADEiger #8: Error when using Stream as Data Source with External triggering

rcjwoods, 2017-08-31:
Hi Bruno,
I've noticed that when I set the Data Source to Stream, and acquire a series of images with external triggering (either External Series, or External Enable), I have to press Stop to end the 'Collecting' status.  This changes the Acquire Status to 'Done', and the Detector State returns 'Aborted'.  The next acquisition causes an ioc error, "StreamApi::readToken<string>: unable to find 'header_detail' token" and "eigerDetector::streamTask: failed to get header packet".

This is different from using Internal triggering. After acquiring an internally triggered series, again using Stream, I press the 'Stop' button, and the Detector State reports 'Idle'. No errors are produced by the next acquisition. So one work-around solution I have is to acquire an Internal Series at the end of each External Series to (in some sense) reset the detector state or clear the ioc error.

I've also found that when using the FileWriter as the Data Source, after sending 'Stop', the Detector State reports 'Aborted', but no errors are produced by the next acquisition, so maybe the Detector State reported is not so relevant to this question.

I hope that makes sense!
Thanks,
Russ

ps. I'm using the up-to-date ADEiger-dev branch.

brunoseivam, 2017-08-31:
Hi Russ,

I can't reproduce it here. If I had to guess, I'd say that at the beginning of the acquisition `streamTask` is probably getting a stray packet indicating the end of the _previous_ acquisition. Here's a patch that makes `getHeader`ignore unexpected packets. Could you try it and see if it solves your issue? 

https://gist.github.com/brunoseivam/2cdac5d63a7d856a95c0d05b0ae61164

Regarding the "Aborted" vs "Idle", if  `streamTask` doesn't get the last packet indicating the end of the acquisition, then it will report that it didn't complete and the driver will say "Aborted" (even if you got all frames). I think a future improvement that could be made is to make `streamTask` aware of the expected number of frames.

External triggers are currently tricky because there's no way to tell when the detector is done. Dectris implemented a `wait` command that doesn't quite work in the current firmware version, but they promised it would be fixed by next version.

rcjwoods, 2017-08-31:
Hi Bruno

Thanks so much for the patch - that fixes my problem! 
The ioc now reports "eigerDetector::streamTask: got stray packet, ignoring" at the beginning of an external trigger acquisition, but it does not prevent subsequent acquisitions. 

Russ

commit abdcd48d (2017-08-31): Stream: ignore non-header packets at the start of the stream. Fixes #8
closed 2017-08-31 by brunoseivam