areaDetector/ADEiger #12: Request - Alarm when local storage running out

johnsinsheimer, 2018-04-17:
I was working with some users and we ran into the unfortunate problem of not using auto-remove of locally stored files, and then running out of local storage.   Could some PVs alarm, maybe on the acquire PV or others that would be readily seen, when the local storage drops below some threshold?  Perhaps minor alarm when less than 10 GB, major alarm when less than 1 GB, or similar.

We found when local storage runs out, the detector will still happily send images to the IOC, but files are no longer saved externally.  We needed to clear local files, and then needed to reboot the DAQ and IOC in order for files to be saved externally again.

brunoseivam, 2018-04-17:
We can add something like:

    field(LOLO, 5)
    field(LLSV, "MAJOR")
    field(LOW,  10)
    field(LSV,  "MINOR")

To

https://github.com/areaDetector/ADEiger/blob/7c6a9596776639a54dbc7e8951ee2f4f9cc70d68/eigerApp/Db/eiger.template#L234-L244

Could you test it locally and then perhaps open a pull request?

Thanks

johnsinsheimer, 2018-04-17:
Thanks Bruno.  We have a shutdown period in about 1 week - I should have time to test this 1-3 weeks from now.