areaDetector/ADEiger #16: Stream frame rate seems impossibly high!

MarkRivers, 2018-08-29:
I am doing some performance measurements using the Stream (ZeroMQ) interface for the ADEiger IOC.

My IOC is running on a Centos7 Linux system.  This system is currently connected to the Eiger server machine by 1Gbit Ethernet.

I see the AcquireTime and Acquire period to 400 microseconds.  This means I should be getting 2500 frames/s.  I set NumImages to 10000, so the entire acquisition should take 4 seconds.

To test this I ran camonitor on the Acquire PV (to see when acquisition started and stopped), and on the Pva1:ArrayRate_RBV PV (to see the rate at which this plugin was being called).
```
corvette:~/support>camonitor 13EIG1:cam1:Acquire 13EIG1:Pva1:ArrayRate_RBV
13EIG1:cam1:Acquire            2018-08-29 07:47:40.673533 Done
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:37.141769 0
13EIG1:cam1:Acquire            2018-08-29 07:48:41.203729 Acquire
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:42.141749 2099
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:43.141787 2496
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:44.141787 2502
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:45.141782 2505
13EIG1:cam1:Acquire            2018-08-29 07:48:45.743706 Done
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:46.141763 398
13EIG1:Pva1:ArrayRate_RBV      2018-08-29 07:48:47.141814 0
```

So Acquire changed to Done 4.5 seconds after it went to Acquire, about 0.5 seconds longer than theoretical.  The PVA plugin ArrayRate is indeed reaching 2500 frames/s.

The problem is that I don't understand how this is physically possible.  The detector is ~0.5M pixels.  At this exposure time the data is UInt16, i.e. 2 bytes per pixel.  Each frame is thus ~1MB, and the total data rate is thus 2500 MB/s.  But 1Gbit Ethernet can only support 125 MB/s, or 20 times less than what I observe.

I thought the ZeroMQ stream was uncompressed, but this must not be true?  Is the ZeroMQ stream compressed, and if so what compression mechanism does it use?

brunoseivam, 2018-08-29:
Hi Mark,

Yes, it is compressed, and the compression is exceptionally good if you're taking empty shots (no beam). Each frame in that case was a few KB (for the Eiger 16M).

As far as I remember, the stream uses the same compression configured for FileWriter. I am pretty sure the compression being used is lz4, the only one I have integrated into ADEiger (`eigerApp/lz4Src`). If you try to use something else, the driver will complain:

https://github.com/areaDetector/ADEiger/blob/0af9b34701c3f3ebbd9f043b65c1a3718a62b963/eigerApp/src/eigerDetector.cpp#L667-L669

MarkRivers, 2018-08-29:
Thanks.  I just figured that out by looking at the source code!

closed 2018-08-29 by MarkRivers