areaDetector/ADEiger #19: port STATSx_TS not found

lcdesilva, 2018-09-14:
Hi All,

I have successfully compiled ADEiger R2-5 on EPICS7. Now I am getting the following message when running the IOC. 

devAsynFloat64Array::initCommon, 13EIG1:Stats1:TSMinValue connectDevice failed asynManager:connectDevice port STATS1_TS not found

It repeats for all TS variables (TSMinX etc) and all STATS variables.

Loading the plugin went just fine. See below.
# Create 5 statistics plugins
NDStatsConfigure("STATS1", 20, 0, "EIG", 0, 0, 0, 0, 0, 5)
dbLoadRecords("NDStats.template",     "P=13EIG1:,R=Stats1:,  PORT=STATS1,ADDR=0,TIMEOUT=1,HIST_SIZE=256,XSIZE=1030,YSIZE=1065,NCHANS=2048,NDARRAY_PORT=EIG")

Any suggestions on how to get past this issue.

Thank you,
Chanaka

GDYendell, 2018-09-14:
This is to do with the change in ADCore 3-3 to NDTimeSeries. These *_TS instances used to be created by the stats plugin, but now the plugin and database need to be created separately. @MarkRivers updated the example for this here https://github.com/areaDetector/ADCore/commit/99c71bc9d0dcee1237c9196ed498f8c2b541ab6c.

lcdesilva, 2018-09-14:
Thank you @GDYendell.

closed 2018-09-14 by lcdesilva