areaDetector/ADEiger #26: ADEiger support for EIGER2?

cmkewish, 2019-03-06:
Hi Bruno, Mark,

I noticed for the original EIGER that there are available API commands "ABORT" and "CANCEL" for stopping exposure series in progress, which don't seem to be supported in areaDetector.

Also, I will be receiving an EIGER2 at the end of the month at the Australian Synchrotron, and I have seen that Dectris added features such as a second energy threshold to output 2 simultaneous images.
Wondering if there has been any discussion about support for EIGER2 in areaDetector: do you have plans for including EIGER2 into ADEiger, or creating a new plugin, at this stage?

Thanks,
Cameron Kewish

cmkewish, 2019-03-06:
Thanks Bruno. Your choice of `abort` makes good sense, thanks for the explanation.

The API commands are the same for EIGER2 for the most part, so I would expect that we can operate the new detector with basic functionality using ADEiger as we have it installed. [I believe the Monitor interface has slightly changed in that they now support both TIFF and NumPy array output and the image type has to be specified in the image request. This has at the moment broken our 'online' viewer applications.]

GDYendell, 2019-03-07:
Hi All,

We are currently running Eiger2s at DLS with this branch: https://github.com/dls-controls/ADEiger/commits/1.8

The 1.8 API has very much been in flux since we got the detector in September. There are currently still some workarounds that we are waiting on updates to be able to take out, so it is not something that can be merged yet. I can tidy up this branch and rebase on the latest master for @brunoseivam and @MarkRivers to take a look at. There are also a few fundamental differences that I believe Dectris are not planning to rectify, so I think it will need some 1.8 specific templates and some version checking in the driver.

brunoseivam, 2019-03-07:
@GDYendell do you know if they plan to release 1.8 for the older Eigers? If so, we can make 1.8 a requirement for using ADEiger. That was my approach in the past when 1.5 / 1.6 was in flux and I didn't want to maintain feature gates in the code.

GDYendell, 2019-03-12:
That would make it much easier. We get the feeling that they are not planning on backporting 1.8 to Eiger1. They are visiting DLS next week so we can ask about that.

GDYendell, 2019-03-15:
See #27 for a WIP PR for 1.8 support.

GDYendell, 2019-05-20:
Just pushed some more updates to the 1.8 PR after rebasing on the latest master.

- A new bit_depth_type parameter has been added
- A load of parameters were added back in, mostly detector metadata
- The access_mode parameter metadata has been fixed, so it doesn't have to assume everything is RW anymore.

cmkewish, 2019-05-22:
@GDYendell - thanks Gary, I have installed that and all seems to be happy. I don't use most of those metadata but it's nice not to have as many white boxes on the MEDM screen :+1:

cmkewish, 2019-05-23:
Just wondering what's the reason for changing `$(P)$(R)FWNamePattern` to a waveform that reports a list of integers representing ASCII character values, rather than leaving it as a string?

GDYendell, 2019-05-23:
I think Mark did that so that the string could be longer than 40 characters.

MarkRivers, 2019-05-23:
That is correct. What command is showing it as a list of integers, rather than as a string?

cmkewish, 2019-05-23:
Understood: thanks for the clarification Mark.

The MEDM window for ADEiger was showing "115" in the text entry field for FWNamePattern when I started the ioc. The readback value was 's'. The default startup value on Eiger initialize should be 'series_$id', so perhaps only the first character was making it into the PV?

If I enter any string into the text entry field to change FWNamePattern, there is a red error pop-up window. But, entering an integer works, say entering 100 results in a readbackvalue of 'd'. Changing 100 to 102 switches the readbackvalue to 'f'.

From the linux commandline: `caget SR05ID01EIG1:cam1:FWNamePattern' gives me a list of 256 space-separated integers, mostly zeros.

I've rolled back to the previous PR WIP:1.8 which I had installed, for now.

MarkRivers, 2019-05-24:
>From the linux commandline: caget SR05ID01EIG1:cam1:FWNamePattern gives me a list of 256 space-separated integers, mostly zeros.

You need to do
caget -S SR05ID01EIG1:cam1:FWNamePattern

The -S tells caget to treat it as a string.

areaDetector uses waveform records for all file paths  and file names, and they work fine with medm.

cmkewish, 2019-05-24:
Thanks Mark -- I've changed out beamline scan system to write waveforms. 

For the ADEiger MEDM window I needed to add a `format="string"` line to get the FWNamePattern text entry box to take and display user input as strings:

```(base) [ics@SR05ID01EIGER02 adl]$ pwd
/beamline/epics/support/areaDetector/ADEiger/eigerApp/op/adl
(base) [ics@SR05ID01EIGER02 adl]$ diff eigerFileWriter.adl eigerFileWriter.adl.orig 
151d150
< 	format="string"
```

MarkRivers, 2019-05-24:
> For the ADEiger MEDM window I needed to add a format="string" line to get the FWNamePattern text entry box to take and display user input as strings:

Sorry, yes that is needed.  I have pushed a new version that fixes that.

cmkewish, 2019-07-02:
For information: I commented in the PR WIP:1.8 that the IOC crashes if I turn on the Monitor mode using the ADEiger MEDM window.
https://github.com/areaDetector/ADEiger/pull/27#issuecomment-507073044

Also, I notice that it take about 30 s exactly to arm the detector every time, where from the RESTful API it usually only takes 30 s on the first arm after INITIALIZE, whereas successive ARMs with the same configuration only take 1 s.

MarkRivers, 2019-07-02:
> For information: I commented in the PR WIP:1.8 that the IOC crashes if I turn on the Monitor mode using the ADEiger MEDM window.
#27 (comment)

> Also, I notice that it take about 30 s exactly to arm the detector every time, where from the RESTful API it usually only takes 30 s on the first arm after INITIALIZE, whereas successive ARMs with the same configuration only take 1 s.

Just to be clear:
- What version of ADEiger are you running?
- What firmware do you have?
- What model Eiger do you have?

cmkewish, 2019-07-02:
I'm running ADEiger version with changes for API 1.8.0 included in "pull request WIP 1.8" from @GDYendell at Diamond
Firmware = eiger_fw_version:  "integration-19.6.21.54194-ga4b949e17"
Model: EIGER2 X 1M

MarkRivers, 2019-07-02:
Are you able to run your detector with the master branch?

If so, do the 2 issues you mentioned above occur on the master branch of ADEiger, or only on the PR?

jlmuir, 2019-12-05:
What's the status of this issue?  I'm planning to use the ADEiger driver to control an EIGER2 X 9M. Should I use the master branch or the latest release?

MarkRivers, 2020-12-29:
This issue is getting obsolete.  The master branch should now support both Eiger1 and Eiger2, and support nearly all of the functionality of the Eiger2 (dual energy thresholds, etc.).

I am going to close this issue.  If there are problems with the master branch please open a new issue.

closed 2020-12-29 by MarkRivers