areaDetector/ADEiger #31: Acquisition Period and Exposure Time fields are not sent to the detector upon a restart

JimboMonkey, 2020-02-06:
Whenever the detector is power cycled, or new firmware is installed, it defaults to an acquisition period of 1 second.  When the Eiger EPICS IOC starts, autosave restores the previously used value for this field (usually something much smaller), and updates the GUI, but doesn't actually 'send' the value to the detector.  

This means that if you then start a data collection it looks like the detector is ignoring the settings and running slowly, when really it is just running at its default value because it hasn't received any other value.  Hitting Enter on the acquisition period and exposure time edm fields causes the desired value to be written and everything works as expected when a collection is started.

I believe this is related to the way autosave can restore values without processing them.

MarkRivers, 2020-12-29:
@JimboMonkey I just tried to reproduce this problem using the master branch on an Eiger2 500K.  I stopped the IOC, and the autosaved values of AcquireTime=0.01, AcquirePeriod=0.01, NumImages=10, NumTriggers=1, TriggerMode=Internal Series. 

I then used the Web interface to the detector to Initialize the detector.  I cannot power-cycle because  am working from home, but I believe that Initialize should be sufficient.  I then started the IOC. The only thing I did in the GUI was press the Start button to set Acquire=1.  It acquired the images at 100Hz, not a 1Hz.

> I believe this is related to the way autosave can restore values without processing them.

autosave never process the records.  However, these records are defined in ADBase.template as follows:
```
record(ao, "$(P)$(R)AcquireTime")
{
   field(PINI, "YES")
   field(DTYP, "asynFloat64")
   field(OUT,  "@asyn($(PORT),$(ADDR=0),$(TIMEOUT=1))ACQ_TIME")
   field(PREC, "3")
   field(VAL,  "1.0")
   info(autosaveFields, "VAL")
}

record(ao, "$(P)$(R)AcquirePeriod")
{
   field(PINI, "YES")
   field(DTYP, "asynFloat64")
   field(OUT,  "@asyn($(PORT),$(ADDR=0),$(TIMEOUT=1))ACQ_PERIOD")
   field(PREC, "3")
   field(VAL,  "0")
   info(autosaveFields, "VAL")
```

eigerBase.template (previously eiger.template) adds these lines:
```
record(ao,"$(P)$(R)AcquireTime") {
    field(VAL, "1")
}

record(ao,"$(P)$(R)AcquirePeriod") {
    field(VAL, "1")
}
```
So it changes the default time from 0 to 1 second for each.

Note that both records have PINI=YES.  This means that the record will process at iocInit.  This processing happens after autosave has restored the value to the record, which will override the default of 1 second.

If it still happens for you with the master branch please tell me the exact sequence of operations that will reproduce it.

MarkRivers, 2022-05-11:
I believe this problem was the same as #58.  @JimboMonkey please feel free re-open this issue if this is not the case.

closed 2022-05-11 by MarkRivers