areaDetector/ADEiger #36: API 1.8 compatibility

antoninomiceli, 2020-07-17:
Hi, 

I think the latest ADEiger release is not compatible with Dectris Eiger API 1.8. They are starting to ship with this version. 

I think there is a fork working on this. https://github.com/dls-controls/ADEiger/tree/1.8

What's the schedule for merging into the main branch?

Thanks
Nino

cmkewish, 2020-07-17:
Hi @antoninomiceli 

See this pull request thread:

https://github.com/areaDetector/ADEiger/pull/30

MarkRivers, 2020-12-15:
The new 1.6_and_1.8 branch supports the 1.8 API.  It should also work with the 1.6 API, though that still needs testing.

You can use that branch until it is merged with master.

closed 2020-12-15 by MarkRivers

antoninomiceli, 2020-12-15:
Thanks! We will try it out on our Eiger1 X 500k which is running the 1.8 API!

MarkRivers, 2020-12-15:
Currently the detector configuration command is passed an argument which is 1 for Eiger1 and 2 for Eiger2.  But I think this flag really means which API is being used (1=1.6, 2=1.8).  The code will need to be improved to support:
- Which API (currently 1.6 or 1.8)
- What model series (currently Eiger1 or Eiger2)
- What firmware version (e.g. 2020.1, 2020.2, etc.)

I think this should be done via the Simplon API early in the constructor, assuming all of this information is available via the Simplon API.