areaDetector/ADEiger #41: Setting photon_energy on Eiger2 returns null

MarkRivers, 2021-01-01:
Setting photon_energy on Eiger2 returns null, rather than a list of parameters that have changed.  This generates an error, and prevents the driver from reading the new value, and the values of other parameters that changed as a result, such as the thresholds.

Not clear if this problem always occurs, or only when the server is in a specific state.

MarkRivers, 2021-01-01:
I tested this issue as follows:

- Using the Web page interface I did Applications/Detector/Restart to restart the detector application.
- Using the Web page interface I did System/Initialize to initialize the detector.
- Used curl to read the value of photon_energy and threshold/1/energy
```
>curl http://164.54.160.41/detector/api/1.8.0/config/photon_energy
{"access_mode":"rw","unit":"eV","value":8041.000000000,"value_type":"float"}
>curl http://164.54.160.41/detector/api/1.8.0/config/threshold/1/energy
{"access_mode":"rw","unit":"eV","value":4020.500000000,"value_type":"float"}
```
- The photon energy was 8041 and the threshold energy was 4020.5.
- Used curl to set the photon_energy to 20000.
```
>curl http://164.54.160.41/detector/api/1.8.0/config/photon_energy -d 20000
null
```
- Note that it returned "null" rather than a list of other parameters that had changed.  Why?
- Used curl to read the value of photon_energy and threshold/1/energy again.
```
>curl http://164.54.160.41/detector/api/1.8.0/config/photon_energy
{"access_mode":"rw","unit":"eV","value":20000.000000000,"value_type":"float"}
>curl http://164.54.160.41/detector/api/1.8.0/config/threshold/1/energy
{"access_mode":"rw","unit":"eV","value":10000.000000000,"value_type":"float"}
```
- Note that the value of 20000 was successfully written, and the threshold energy was set to 10000, as expected.

MarkRivers, 2021-01-09:
Dectris was able to reproduce this problem.  

Thanks to help from Zachary Brown we were able to figure out that the following appears to be the minimal sequence of operations to restart the Eiger DCU while having a "put" to photon_energy correctly return the list of modified parameters.

Using the Eiger Web page interface do the following:
- Applications/Detector/Restart to restart the detector application
- System/Initialize to initialize the detector.
-  Applications/calibration-xxx/Restart to restart the calibration application
- System/Initialize to initialize the detector again

After doing those steps the curl command to set the photon_energy correctly returns the list of modified parameters:
```
>curl http://164.54.160.41/detector/api/1.8.0/config/photon_energy -d 10000
["element","flatfield","photon_energy","threshold/1/energy","threshold/1/flatfield","threshold/2/energy","threshold/2/flatfield","threshold_energy","wavelength"]
```
The EPICS IOC also correctly reads back the modified energy thresholds after the photon_energy is set.

Dectris should be fixing this issue in a coming release.

tguruswamy, 2024-11-14:
This issue has been resolved in the 2024.1 firmware release from Dectris.