areaDetector/ADEiger #56: Aborting acquisition with long acquire time can hang, require IOC reboot.

MarkRivers, 2022-04-25:
Stopping an acquisition with long acquire times like 5 seconds can hang the system.  This happens in Internal Series and External Series mode.  The symptom is that StausMessage_RBV will say "Acquisition aborted" rather than "Ready" after the Stop button is pressed while it is acquiring a series.  When pressing Start to try to start another acquisition the following errors appear at the IOC:
```
StreamApi::getHeader: wrong header type, htype=dimage-1.0
2022/04/25 16:38:30.944 eigerDetector::streamTask: got stray packet, ignoring
StreamApi::getHeader: wrong header type, htype=dimage_d-1.0
2022/04/25 16:38:30.944 eigerDetector::streamTask: got stray packet, ignoring
StreamApi::getHeader: failed to parse JSON, data=
2022/04/25 16:38:30.945 eigerDetector::streamTask: failed to get header packet
```
The system will no longer acquired correctly.  It requires restarting the IOC.  It is not necessary to restart the Eiger server.

This issue is observed on the Eiger2 detectors with firmware release-2020.2.2.  It happens nearly every time an acquisition is stopped on an Eiger 9M when the AcquireTime is 2 seconds or more.  It happens much less frequently on an Eiger 1M, perhaps 5% of the time.

byeongdu, 2022-04-25:
This issue is identical to #54

MarkRivers, 2022-05-03:
StatusMessage_RBV is set to "Acquisition aborted" here: 
https://github.com/areaDetector/ADEiger/blob/011daeef70ba38be951409b366767298a87545d3/eigerApp/src/eigerDetector.cpp#L1001
That happens when adStatus == ADStatusAborted && success=false.

The first condition is expected, because we aborted the acquisition.  The question is why is success==false?

success is set to true here: https://github.com/areaDetector/ADEiger/blob/011daeef70ba38be951409b366767298a87545d3/eigerApp/src/eigerDetector.cpp#L963

Because we were not using the Filewriter during these tests, the only way that success can be set to false is in this line:
https://github.com/areaDetector/ADEiger/blob/011daeef70ba38be951409b366767298a87545d3/eigerApp/src/eigerDetector.cpp#L989  That happens when mStreamComplete is false after calling mStreamDoneEvent.wait(). 

I am not sure how the problem is happening.  mStreamComplete is only set to false at the beginning of acquiring a series.  It is set to true as soon as the streamTask gets the first valid frame in a series.  We are having the problem when aborting a series after the first frame has been received, so how can mStreamComplete be false?

The way to track this down is to enable ASYN_TRACE_FLOW.  There will then be messages printed with the value of mStreamComplete and hopefully we can figure out the problem.

Since there is no power on most APS beamlines until Friday this week this may have to wait until then to debug.

MarkRivers, 2022-05-04:
This was observed to occur on the Eiger2 500K at 8-ID as well.  It seemed to occur about 10-20% of the time with 1 second acquire time and period, which is more frequent than the Eiger2 1M at 13-ID.

MarkRivers, 2022-05-04:
I do understand the problem.  streamTask only sets mStreamComplete when it receives frame.end.  It does not receive that after each frame, it only receives it after the *last* frame.  So mStreamComplete it still false when acquiring() returns false because a series was stopped.  That explains why StatusMessage_RBV is "Acquisition aborted".  It does not change to "Ready" because ADStatus never gets set back to ADStatusIdle.  This is not really a problem.

The real problem is that the next time we try to acquire we get this message:
```
StreamApi::getHeader: wrong header type, htype=dimage-1.0
2022/04/25 16:38:30.944 eigerDetector::streamTask: got stray packet, ignoring
StreamApi::getHeader: wrong header type, htype=dimage_d-1.0
2022/04/25 16:38:30.944 eigerDetector::streamTask: got stray packet, ignoring
StreamApi::getHeader: failed to parse JSON, data=
2022/04/25 16:38:30.945 eigerDetector::streamTask: failed to get header packet
```
I believe I now understand that problem.  It is a side-effect of my fix to the race condition in ADEiger R3-1 and R3-2.  Prior to those fixes the ZMQ socket was created at the beginning of each acquisition, and deleted at the end.  Now the ZMQ socket is created as soon as DataSource is set to Stream, and deleted when it is set to any other than Stream.  

This is causing a problem because the ZMQ socket may still have some stale data in it that is not flushed when a series is aborted.  That stale data is then read at the start of the next acquisition. That is why it is receiving a dimage message when it is expecting a dheader message.

The solution is either to flush the ZMQ socket by reading until a timeout of at least the AcquirePeriod expires, or close and re-open the ZMQ socket.

PR #59 (2022-05-06): Fix problem with stopping a series
This is an attempt to fix #56 

It does the following:
- Improves the legbility of streamTask by using switch statements and explicitly handling all values of stream_err.
- If aquisition is stopped then streamTask deletes and recreates mStreamAPI.  This closes and reopens the ZMQ socket, which flushes any existing data from the aborted scan.
- If an acquisition is stopped then StatusMessage_RBV will always be set to "Acquisition aborted".  Previously it was only set to that value if the frame.end was not received.  That was confusing.

This PR has not been te [... cut]
closed 2022-05-06 by MarkRivers