areaDetector/ADEiger #64: Stream mode with CompressionAlgo=None and Decompress=No gives "unknown encoding <"

tguruswamy, 2024-01-09:
Logically, Stream mode should work with CompressionAlgo=None and StreamDecompress=No, but this case (encoding "<") is not handled in eigerDetector.cpp -- an "unknown encoding" error results.

GDYendell, 2024-01-18:
Hi @tguruswamy. I had to confirm this with Dectris because I was under the impression that it was not possible to get uncompressed data through the stream interface, but this is not the case.

I think the driver possibly should support this rather than giving an error as it does currently. However, it is strongly recommended to only use stream with bslz4 compressed data and they give no guarantees about the performance you might get otherwise. I quote from a support email with Dectris:

> Please be aware that no performance testing is done with no compression, so it may not work depending on the detector type and framerate. We don't recommend using no compression in Stream, and full performance is only guaranteed with bslz4.

Is there are reason you don't want to use compression? Is this just for testing or would you want to put this into production?

tguruswamy, 2024-01-19:
Yes, your understanding is correct -- for full speed acquisition we are certainly using compression. We're aware of Dectris's recommendations.

The use case for uncompressed data is more for prototyping, and processing by other areaDetector plugins. Consider some kind of automatic beam alignment loop relying on the Stats plugin, for example. While in principle it is possible to handle the Eiger compressed data by inserting a ADCodec plugin:

`ADEiger (BSLZ4) compressed stream -> Codec (decompress mode) -> ROI -> Stats, Overlay, etc. -> PVA/Image/File`

, in our testing Codec is not particularly fast (even with multiple threads) and quickly becomes the bottleneck, not the network transfer from the Eiger.

So for these cases where a very high frame rate is not needed and significant areaDetector plugin processing is wanted we would prefer:

`ADEiger uncompressed stream -> ROI -> Stats, Overlay, etc. -> PVA/Image/File`

, simply to save resources on the IOC computer. And this does work, partially -- with `Decompress=True` and `CompressionAlgo=None` -- but not with `Decompress=False` and `CompressionAlgo=None`. The original motivation of this bug report was that specific inconsistency, which was discovered during testing.

(And we are not using FileWriter mode because we want each frame as it is acquired.)

MarkRivers, 2024-01-19:
> While in principle it is possible to handle the Eiger compressed data by inserting a ADCodec plugin:

You don't need a plugin for this.  ADEiger can decompress the images itself.

tguruswamy, 2024-01-19:
> > While in principle it is possible to handle the Eiger compressed data by inserting a ADCodec plugin:
> 
> You don't need a plugin for this. ADEiger can decompress the images itself.

Indeed, however that is even slower than Codec. I assume because it is happening in the single eiger-stream thread. (For similar reasons we also use Codec over FileHDF5's compression).