areaDetector/ADEiger #65: Support Stream2 CBOR format and multiple threshold frames

tguruswamy, 2024-09-23:
(TODO)
Stream2 with compression is the only supported way to capture data at full speed with multiple photon energy thresholds enabled.

Data structure (CBOR) documentation is available here: https://github.com/dectris/documentation

@MarkRivers suggests handling multiple thresholds as an extra dimension in the NDArray. This is also how Dectris construct HDF files in FileWriter2 mode, saving as a 4-dimensional array (nimages * nthresholds * X * Y). This may require updates to viewer/analysis software to correctly interpret.

MarkRivers, 2024-09-23:
> This may require updates to viewer/analysis software to correctly interpret.

If the ADEiger driver produces an NDArray of dimensions NX x NY x 2 then this can be sent to 2 ROI plugins.  One selects the low threshold image and the other selects the high threshold image.  These are sent to 2 separate PVA plugins.  Viewers then don't need to be modified, they can chose which image to display by selecting which PVA plugin to connect to.  If they want to display both images that would require changes.

MarkRivers, 2025-04-02:
I plan to begin working on this soon.  Does anyone know of existing efforts to add Stream2 support to ADEiger?

MarkRivers, 2025-04-23:
I have created a stream2 branch for this work.  It is able to receive and decode the stream2 CBOR zmq messages.  It currently works for single-threshold data with compression=None or bslz4.  It crashes with compression=lz4 because it appears that stream2 is using a different lz4 compression than stream.

MarkRivers, 2025-06-15:
This is now implemented, and will appear in the R3-6 release.

closed 2025-06-15 by MarkRivers